unitreerobotics/unitree_ros #2: Package Built Error

tahir1069, 2020-11-18:
I am trying to build unitree_ros package but no success. Here are the steps I followed for the installations:

`sudo apt install libboost-all-dev`
`sudo apt install liblcm-dev`
`sudo apt install cmake`
`sudo apt-get install ros-melodic-controller-interface  ros-melodic-gazebo-ros-control ros-melodic-joint-state-controller ros-melodic-effort-controllers ros-melodic-joint-trajectory-controller`

Now I believe all the dependencies would have been installed to latest stable release.
Now I exactly followed the same procedure as in the Readme:

1. To build unitree_legged_sdk
```
git clone https://github.com/unitreerobotics/unitree_legged_sdk
cd unitree_legged_sdk
mkdir build
cd build
cmake ..
make
```

2.  To build aliengo sdk:
```
git clone https://github.com/unitreerobotics/aliengo_sdk
cd aliengo_sdk
mkdir build
cd build
cmake ..
make

```
3. Creating Catkin_ws and adding unitree_ros
```
mkdir -p catkin_ws/src
cd catkin_ws/src
git clone https://github.com/unitreerobotics/unitree_ros
cd ..
catkin build
```

Now all of the below environment variable and source commands exist in  _.bashrc_

```
source /opt/ros/melodic/setup.bash
source /usr/share/gazebo-9/setup.sh
source ~/catkin_ws/devel/setup.bash
export ROS_PACKAGE_PATH=~/catkin_ws:${ROS_PACKAGE_PATH}
export GAZEBO_PLUGIN_PATH=~/catkin_ws/devel/lib:${GAZEBO_PLUGIN_PATH}
export LD_LIBRARY_PATH=~/catkin_ws/devel/lib:${LD_LIBRARY_PATH}
export UNITREE_LEGGED_SDK_PATH=~/unitree_legged_sdk
export ALIENGO_SDK_PATH=~/aliengo_sdk
export UNITREE_PLATFORM="amd64"
```
After the build, _unitree_legged_real_ package is not build giving the error:
```

make[2]: *** No rule to make target 'CMakeFiles/lcm_server_3_1.dir/build'.  Stop.
make[1]: *** [CMakeFiles/lcm_server_3_1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target 'CMakeFiles/lcm_server_3_2.dir/build'.  Stop.
make[1]: *** [CMakeFiles/lcm_server_3_2.dir/all] Error 2
/home/mbs/catkin_ws/src/unitree_ros/unitree_legged_real/src/exe/walk_mode.cpp:13:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/walk_lcm.dir/src/exe/walk_mode.cpp.o] Error 1
make[1]: *** [CMakeFiles/walk_lcm.dir/all] Error 2
/home/mbs/catkin_ws/src/unitree_ros/unitree_legged_real/src/exe/velocity_mode.cpp:13:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/velocity_lcm.dir/src/exe/velocity_mode.cpp.o] Error 1
make[1]: *** [CMakeFiles/velocity_lcm.dir/all] Error 2
/home/mbs/catkin_ws/src/unitree_ros/unitree_legged_real/src/exe/torque_mode.cpp:12:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/torque_lcm.dir/src/exe/torque_mode.cpp.o] Error 1
make[1]: *** [CMakeFiles/torque_lcm.dir/all] Error 2
/home/mbs/catkin_ws/src/unitree_ros/unitree_legged_real/src/exe/position_mode.cpp:13:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o] Error 1
make[1]: *** [CMakeFiles/position_lcm.dir/all] Error 2
make: *** [all] Error 2
cd /home/mbs/catkin_ws/build/unitree_legged_real; catkin build --get-env unitree_legged_real | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -

```

All the environment variables are correctly set but still it is not able to find _unitree_legged_sdk_. 
```
rosPC:~$ printenv | grep "UNITREE"
UNITREE_PLATFORM=amd64
UNITREE_LEGGED_SDK_PATH=/home/mbs/unitree_legged_sdk

```
Is there anything which I am missing or something else needs to be done?
Any leads/help will highly be appreciated.

closed 2020-11-18 by tahir1069

SSmarshal, 2021-01-29:
Hello, how do you solve your problem, I come up with the same problem?

tahir1069, 2021-01-31:
Hi
I was able to solve the problem by unintsalling everything and than again installing respective versions.

SSmarshal, 2021-01-31:
Hello, sorry to interrupt, but may I ask the version of gazebo you used? Is it gazebo8 or other version? I have already built the package but when I followed the demo and rosrun unitree_controller unitree_servo. Nothing just happened. I am using ros melodic + gazebo9(It seems that ubuntu18.04 does not support gazebo8 but melodic requires 18.04?)

tahir1069, 2021-01-31:
Well yes you are right that's the problem for Gazebo version. You can just path of Gazebo9 when setting up the environment variables. I tested with this and everything intended shown up.

SSmarshal, 2021-02-01:
Thank you, I could open up the world, however, when I rosrun unitree_controller unitree_servo, the robot could not stand up as expected, and it said that "Controller Spawner couldn't find the expected controller_manager ROS interface." I doubted that it was because the version of the gazebo is not right?

tahir1069, 2021-02-01:
I don't think so that for me it worked well. Can you please suggest a screenshot or log for your commands you run.

SSmarshal, 2021-02-01:
Thank you for replying！The problem I come up with was when I run "roslaunch unitree_gazebo normal.launch rname:=a1 wname:=stairs", the robot and the stair could show up as shown in the following figure:
[image]
however, it said:
"[WARN] [1612175390.412604, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[a1_gazebo/controller_spawner-5] process has finished cleanly
"
Then I run "rosservice list | grep controller_manager", it had the output which seemed ok:
[image]
Then I checked the namespace of the launch file and the xacro file, the namespace of the controller_manager and the spawner was quite the same:
[image]
[image]
But it still said"Controller Spawner couldn't find the expected controller_manager ROS interface.
"
And I am confused about how to fix it. Thank you!

SSmarshal, 2021-02-01:
By the way，may I have a look at your bashrc of the relative settings， thank you

SSmarshal, 2021-02-01:
I have solved the problem, the uri of the stairs.world was wrong, sorry to interrupt, and thank you!