unitreerobotics/unitree_ros #5: When high level mode will be available in gazebo?

AlexKaravaev, 2020-12-01:
Hi, we have one of A1 robots and we need some simulation testing to run the robot. 

When the walking will be available in gazebo? If not, will it be hard to write this package by our own?

tahir1069, 2021-02-01:
Hi if you are not looking for Gazebo specifically. We have a working example in Webots simulator. You can find the repo [here](https://github.com/MYBOTSHOP/mbs_unitree_ros). We are actively working on it so you can update as well.
Hope this helps.

Zhaiweiwei0, 2021-07-14:
Sorry, we do not have this ROS package，this is a little difficult for beginners.

closed 2021-10-11 by TrivasZhang

itaouil, 2021-11-02:
Hi @edward9503  @AlexKaravaev,

Did you by any chance figure out a way to do high level control in simulation?

AlexKaravaev, 2021-11-03:
@itaouil Hi!

Unfortunetaly no. Seems like it will require to write own control of the joints, which seems to be pretty hard