unitreerobotics/unitree_ros #6: Build Error

dlee640, 2020-12-22:
Hello. I have A1 and I am trying to set up the environment.

Currently, my catkin_ws is named a1_ws and I have src, devel and build folder within it.
Within the src folder, I have included unitree_ros folder and unitree_legged_sdk.

When I try to run catkin_make, I get the following error about unitree_legged_sdk.h
`dlee640@dlee640-ThinkPad-T580:~/a1_ws$ catkin_make -DCATKIN_WHITELIST_PACKAGES=""
Base path: /home/dlee640/a1_ws
Source space: /home/dlee640/a1_ws/src
Build space: /home/dlee640/a1_ws/build
Devel space: /home/dlee640/a1_ws/devel
Install space: /home/dlee640/a1_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/dlee640/a1_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/dlee640/a1_ws/build"
####
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_IMU
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_HighCmd
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_MotorState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_LowState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_LED
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_HighState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_MotorCmd
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_LowCmd
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_Cartesian
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target controller_manager_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_eus
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target actionlib_msgs_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_eus
[  0%] Built target tf_generate_messages_eus
[  0%] Built target tf_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target actionlib_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target tf_generate_messages_py
[  0%] Built target actionlib_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target controller_manager_msgs_generate_messages_eus
[  0%] Built target controller_manager_msgs_generate_messages_nodejs
[  0%] Built target controller_manager_msgs_generate_messages_py
[  0%] Built target controller_manager_msgs_generate_messages_cpp
[  2%] Built target unitree_move_kinetic
[  5%] Built target unitree_external_force
[ 18%] Built target unitree_legged_msgs_generate_messages_lisp
[ 33%] Built target unitree_legged_msgs_generate_messages_cpp
[ 46%] Built target unitree_legged_msgs_generate_messages_py
[ 49%] Built target unitreeFootContactPlugin
[ 52%] Built target unitreeDrawForcePlugin
[ 65%] Built target unitree_legged_msgs_generate_messages_nodejs
[ 79%] Built target unitree_legged_msgs_generate_messages_eus
[ 79%] Built target unitree_legged_msgs_gencpp
[ 81%] Building CXX object unitree_ros/unitree_legged_real/CMakeFiles/velocity_lcm.dir/src/exe/velocity_mode.cpp.o
[ 81%] Built target unitree_legged_msgs_generate_messages
[ 82%] Building CXX object unitree_ros/unitree_legged_real/CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o
[ 84%] Building CXX object unitree_ros/unitree_legged_real/CMakeFiles/walk_lcm.dir/src/exe/walk_mode.cpp.o
[ 85%] Building CXX object unitree_ros/unitree_legged_real/CMakeFiles/torque_lcm.dir/src/exe/torque_mode.cpp.o
[ 88%] Built target unitree_controller
[ 91%] Built target unitree_legged_control
[ 94%] Built target unitree_servo
/home/dlee640/a1_ws/src/unitree_ros/unitree_legged_real/src/exe/walk_mode.cpp:13:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/dlee640/a1_ws/src/unitree_ros/unitree_legged_real/src/exe/torque_mode.cpp:12:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
unitree_ros/unitree_legged_real/CMakeFiles/walk_lcm.dir/build.make:62: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/walk_lcm.dir/src/exe/walk_mode.cpp.o' failed
make[2]: *** [unitree_ros/unitree_legged_real/CMakeFiles/walk_lcm.dir/src/exe/walk_mode.cpp.o] Error 1
CMakeFiles/Makefile2:2862: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/walk_lcm.dir/all' failed
make[1]: *** [unitree_ros/unitree_legged_real/CMakeFiles/walk_lcm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
unitree_ros/unitree_legged_real/CMakeFiles/torque_lcm.dir/build.make:62: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/torque_lcm.dir/src/exe/torque_mode.cpp.o' failed
make[2]: *** [unitree_ros/unitree_legged_real/CMakeFiles/torque_lcm.dir/src/exe/torque_mode.cpp.o] Error 1
CMakeFiles/Makefile2:3231: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/torque_lcm.dir/all' failed
make[1]: *** [unitree_ros/unitree_legged_real/CMakeFiles/torque_lcm.dir/all] Error 2
/home/dlee640/a1_ws/src/unitree_ros/unitree_legged_real/src/exe/velocity_mode.cpp:13:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
/home/dlee640/a1_ws/src/unitree_ros/unitree_legged_real/src/exe/position_mode.cpp:13:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
 #include "unitree_legged_sdk/unitree_legged_sdk.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
unitree_ros/unitree_legged_real/CMakeFiles/position_lcm.dir/build.make:62: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o' failed
make[2]: *** [unitree_ros/unitree_legged_real/CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o] Error 1
unitree_ros/unitree_legged_real/CMakeFiles/velocity_lcm.dir/build.make:62: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/velocity_lcm.dir/src/exe/velocity_mode.cpp.o' failed
make[2]: *** [unitree_ros/unitree_legged_real/CMakeFiles/velocity_lcm.dir/src/exe/velocity_mode.cpp.o] Error 1
CMakeFiles/Makefile2:3140: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/position_lcm.dir/all' failed
make[1]: *** [unitree_ros/unitree_legged_real/CMakeFiles/position_lcm.dir/all] Error 2
CMakeFiles/Makefile2:2953: recipe for target 'unitree_ros/unitree_legged_real/CMakeFiles/velocity_lcm.dir/all' failed
make[1]: *** [unitree_ros/unitree_legged_real/CMakeFiles/velocity_lcm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

What am I doing wrong?

dlee640, 2020-12-23:
This issue has been solved.

closed 2020-12-23 by dlee640

SSmarshal, 2021-01-29:
Hello, I come up with similar problem, may I ask how do you solve your problem?

offjangir, 2021-03-07:
@dlee640  what did you do to solve the error am getting the same error ?

dlee640, 2021-04-08:
Add following to your workspace/src/CMakeLists.txt
```
include_directories(
  include
  ${catkin_INCLUDE_DIRS}
)
```

HDElectronics, 2022-05-28:
> 

I added the instruction, but still not working any other suggestion please ?

wongearth, 2022-09-04:
Same problem... I can not solve with suggest above. Please be more elaborate.

JonasFovea, 2022-09-20:
So I have the same problem.

As far as I understand, the cpp files in the `unitree_legged_real` package (from the unitree_ros_to_real repository) include the unitree_legged_sdk.h file.
This file however is located in the unitree_legged_sdk repository and therefore not in the same package..

So my guess wold be to include this repo somehow in the CMakeLists.txt.

My file structure inside catkin_ws/src:

```
├── unitree_legged_sdk
│   ├── CMakeLists.txt
│   ├── example
│   │   ├── example_joystick.cpp
│   │   ├── example_position.cpp
│   │   ├── example_torque.cpp
│   │   ├── example_velocity.cpp
│   │   └── example_walk.cpp
│   ├── example_py
│   │   ├── example_position.py
│   │   ├── example_torque.py
│   │   ├── example_velocity.py
│   │   └── example_walk.py
│   ├── include
│   │   └── unitree_legged_sdk
│   │       ├── a1_const.h
│   │       ├── aliengo_const.h
│   │       ├── comm.h
│   │       ├── go1_const.h
│   │       ├── joystick.h
│   │       ├── loop.h
│   │       ├── quadruped.h
│   │       ├── safety.h
│   │       ├── udp.h
│   │       └── unitree_legged_sdk.h                                      <== The included file
│   ├── lib
│   │   ├── cpp
│   │   │   ├── amd64
│   │   │   │   └── libunitree_legged_sdk.a
│   │   │   └── arm64
│   │   │       └── libunitree_legged_sdk.a
│   │   └── python
│   │       ├── amd64
│   │       │   └── robot_interface.cpython-38-x86_64-linux-gnu.so
│   │       └── arm64
│   │           ├── robot_interface.cpython-37m-aarch64-linux-gnu.so
│   │           └── robot_interface.cpython-38-aarch64-linux-gnu.so
│   ├── LICENSE
│   └── README.md
└── unitree_ros_to_real
    ├── README.md
    ├── unitree_legged_msgs
    │   ├── CMakeLists.txt
    │   ├── msg
    │   │   ├── BmsCmd.msg
    │   │   ├── BmsState.msg
    │   │   ├── Cartesian.msg
    │   │   ├── HighCmd.msg
    │   │   ├── HighState.msg
    │   │   ├── IMU.msg
    │   │   ├── LED.msg
    │   │   ├── LowCmd.msg
    │   │   ├── LowState.msg
    │   │   ├── MotorCmd.msg
    │   │   └── MotorState.msg
    │   └── package.xml
    └── unitree_legged_real
        ├── CMakeLists.txt
        ├── include
        │   └── convert.h
        ├── ipconfig.sh
        ├── launch
        │   ├── keyboard_control.launch
        │   └── real.launch
        ├── package.xml
        └── src
            └── exe                    <== The directory, from which it is included (multiple files)
                ├── control_via_keyboard.cpp                                   
                ├── example_position.cpp
                ├── example_walk.cpp
                ├── ros_udp.cpp
                ├── state_sub.cpp
                └── twist_sub.cpp

```

Does somebody know how to do this?

JulienRineau, 2022-12-01:
Has someone fix this problem ?

JonasFovea, 2022-12-02:
Hi @JulienRineau !

You have to include the the "include" directory of the SDK in your CMakeLists.txt file:

https://github.com/JonasFovea/unitree_ros_to_real/blob/61857e62a421ea3dd403ee5a7242c724a4c3b19e/unitree_legged_real/CMakeLists.txt#L29-L37

If you have further build errors, you can have a look into my forks of the SDK and this repo, in the verions I'll link below, the branches are working well for me and the Go1 robot.

https://github.com/JonasFovea/unitree_legged_sdk/tree/fix
https://github.com/JonasFovea/unitree_ros_to_real/tree/chip

If you have further questions, feel free to ask!

JulienRineau, 2022-12-02:
Thank you @JonasFovea for your answer.
I have this error when I try to build the [a1_cpp](https://github.com/ShuoYangRobotics/A1-QP-MPC-Controller/tree/main/src/a1_cpp) package. 

Even after making the changes you mentioned (I tried both in _unitree_legged_real_ and my _a1_cpp_ package) I still have the following error:
```
[ 82%] Building CXX object CMakeFiles/hardware_a1_lib.dir/src/HardwareA1ROS.cpp.o
In file included from /home/parallels/capstone/workspace/src/a1_cpp/src/HardwareA1ROS.cpp:5:
/home/parallels/capstone/workspace/src/a1_cpp/src/HardwareA1ROS.h:40:10: fatal error: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory
   40 | #include "unitree_legged_sdk/unitree_legged_sdk.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/hardware_a1_lib.dir/build.make:63: CMakeFiles/hardware_a1_lib.dir/src/HardwareA1ROS.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:300: CMakeFiles/hardware_a1_lib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
<== Failed to process package 'a1_cpp': 
  Command '['/home/parallels/capstone/workspace/devel_isolated/unitree_legged_msgs/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2.

Reproduce this error by running:
==> cd /home/parallels/capstone/workspace/build_isolated/a1_cpp && /home/parallels/capstone/workspace/devel_isolated/unitree_legged_msgs/env.sh make -j4 -l4

Command failed, exiting.
```
Am I doing something wrong ?

JonasFovea, 2022-12-05:
@JulienRineau could you give us the link to your fork of the mentioned repository?
Or at least the content of your CMakeLists.txt and your directory tree?

I'm really not sure, what causes your problem.
But I'm definitely no expert on CMake and catkin...