unitreerobotics/unitree_ros #10: Node::Advertise(): Error advertising a topic. Did you forget to start the discovery service?

SSmarshal, 2021-01-30:
Hi，I am using gazebo9 + ros melodic(it seems that ubuntu18.04 does not support gazebo8 but ros melodic needs ubuntu18.04).
And I launched the world successfully but when I run the following code:
`"rosrun unitree_controller unitree_servo"`
Nothing happens in gazebo.
And the terminal says Node::Advertise(): Error advertising a topic. Did you forget to start the discovery service?
And I do not know how to fix it?
Any help will be appreciated! Thanks in advance!

Zhaiweiwei0, 2021-07-14:
Hi, please provide more information on this issue.

TrivasZhang, 2021-10-11:
@SSmarshal  
Any progress？

amit-420, 2021-11-13:
@SSmarshal could explain more how you fixed the problem, I am also facing similar issue.
``` Node::Advertise(): Error advertising topic [default/turtlebot3_burger/base_footprint/camera/image]. Did you forget to start the discovery service? ```

mgrallos, 2022-03-10:
Facing the same issues just by running gazebo and saving a world. Any updates on how to solve this ?

BartVanderHaagen, 2022-04-30:
I auto generated an design of a spiderbot from using fusion2urdf 
when lauching the gazebo.launch it starts just fine only there is no control over the joints

`Topic [/data://world/default/model/spiderbot_step3/plugin/control/joint_cmd] is not valid.`

any help would be nice

DragonflyRobotics, 2023-01-16:
@BartVanderHaagen I am facing a similar issue.

```
Node::Advertise(): Error advertising topic [/zeta_urdf/joint_cmd]. Did you forget to start the discovery service?
```

scotgopal, 2023-06-30:
> > @SSmarshal Any progress？
> 
> Sorry to reply late, I made some mistakes in the file staris.world.

What was the mistake and what's the fix?

CiaranYoung, 2023-12-20:
`Node::Advertise(): Error advertising a topic. Did you forget to start the discovery service?
`the same problem

SSmarshal, 2023-12-20:
Sorry guys for a long time no reply, something happened to me and it's been a long time since I made a quadruped robot. I think it was a very small mistake, like a misspelling of a word.

SSmarshal, 2023-12-20:
I have contacted my former partner. There may be a memory error, but it seems that the filename was mistyped-up when a file was called from the staris.world.