unitreerobotics/unitree_ros #7: ROS 2 Support?

SteveMacenski, 2021-01-30:
Hi,

I have an A1 with the expressed goal of making it work with the new and improved ROS 2 navigation2 system. Do you have drivers that work with ROS 2 foxy or any of the ROS2 distributions?

dlee640, 2021-06-29:
Hello @SteveMacenski , it's nice to see you here (I am familiar with your work on spatio temporal voxel filter!). I have been working on this exact task recently. Please let me know if your team is working on this, or looking for someone to collaborate. Thank you.

SteveMacenski, 2021-06-29:
I've essentially given up on it and am waiting for someone else to do it. I love the Unitree hardware, the robot itself is great, but the software support is terrible and a ton of false advertising

ZKBian, 2021-07-14:
Hello, until now we does not work on ROS2, and does not plan to support ROS2 recently.

closed 2021-07-14 by ZKBian

MAVProxyUser, 2022-11-28:
It is unfortunate Unitree is so non-responsive to requests like this. Github is festering with questions that remain unanswered.

MAVProxyUser, 2022-11-28:
This seems to address the question at hand? https://github.com/unitreerobotics/unitree_ros2_to_real/blob/main/ros2_unitree_legged_msgs/CMakeLists.txt

JuloWaks, 2023-05-04:
Any updates? Is ROS2 officially supported?

kitare17, 2025-08-26:
Is ROS2 officially supported?