unitreerobotics/unitree_ros #12: Cannot install unitree_legged_real

hansenmaster, 2021-03-04:
Hello, I have problem when doiing catkin_make on unitree_legged_real:

[ 84%] Linking CXX executable /home/hansen/catkin_ws/devel/lib/unitree_legged_real/walk_lcm
/usr/bin/ld: cannot find -lunitree_legged_sdk_
/usr/bin/ld: cannot find -laliengo_comm
collect2: error: ld returned 1 exit status

I have followed the instruction to install unitree_legged_sdk and aliengo_sdk, but it seems that the catkin_make cannot find this library. Does anyone have idea of a solution?

Thanks

Jtang1188, 2021-04-28:
Did you try putting the unitree_legged_sdk and aliengo_sdk folder into the unitree_ros folder? And run it afterwards

ChristyanCruz11, 2021-05-28:
Hello, I have problem when doiing catkin_make on unitree_legged_real:

[ 84%] Linking CXX executable /home/hansen/catkin_ws/devel/lib/unitree_legged_real/walk_lcm
/usr/bin/ld: cannot find -lunitree_legged_sdk_
/usr/bin/ld: cannot find -laliengo_comm

Thats exactly my problem,

Zhaiweiwei0, 2021-07-14:
Hi, maybe you can try the following methods:
1. Install the unitree_legged_sdk and aliengo_sdk under the home directory(/home/username/);
2. Make sure the following exist in your ~/.bashrc:
export UNITREE_LEGGED_SDK_PATH=~/unitree_legged_sdk
export ALIENGO_SDK_PATH=~/aliengo_sdk

hansenmaster, 2021-07-14:
Thanks everyone, I have successfully catkin_make  on unitree_legged_real. 
It seems the solution is as @Zhaiweiwei0 suggests to make sure the directory and path is properly set. I am closing this.

closed 2021-07-14 by hansenmaster