unitreerobotics/unitree_ros #14: 报错

jingxingdaze, 2021-04-08:
unitree_ros里面的unitree_legged_real的CmakeLists，最后的add_dependencies里的{PEOJECT_NAME}打错了，应该是PROJECT，后面都是复制粘贴的所以都错了

alt0216, 2021-04-13:
Hello. Are you able to run any of the controllers: `position_lcm, velocity_lcm, torque_lcm, walk_lcm` after the fix?

jingxingdaze, 2021-04-14:
> Hello. Are you able to run any of the controllers: `position_lcm, velocity_lcm, torque_lcm, walk_lcm` after the fix?
Yes, and I also fix the
 "string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_$ENV{UNITREE_PLATFORM}.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} libaliengo_comm.so lcm)"
from Cmakelists to
 "string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_$ENV{UNITREE_PLATFORM}.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} libaliengo_comm.so lcm)" because of the framework we were using.

jingxingdaze, 2021-04-14:
> Hello. Are you able to run any of the controllers: `position_lcm, velocity_lcm, torque_lcm, walk_lcm` after the fix?

Yes, and I also fix the
"string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_$ENV{UNITREE_PLATFORM}.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} libaliengo_comm.so lcm)"
from Cmakelists to
"string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_arm64.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} libaliengo_comm_arm64.so lcm)" because of the framework we were using.

jingxingdaze, 2021-04-14:
> Hello. Are you able to run any of the controllers: `position_lcm, velocity_lcm, torque_lcm, walk_lcm` after the fix?
Yes, and I also fixed the
"string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_$ENV{UNITREE_PLATFORM}.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} libaliengo_comm.so lcm)"
in Cmakelists to
"string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_arm64.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} libaliengo_comm_arm64.so lcm)" because of the framework we were using.

ZKBian, 2021-07-14:
Thank you for your report! We have fixed this bug.

closed 2021-07-14 by ZKBian