unitreerobotics/unitree_ros #17: Cannot run through the "catkin_make" for building the project at the first time

edward9503, 2021-07-30:
Hi there,

My Ubuntu is 20.04.2 LTS, and Gazebo is 11. When i first time run the catkin_make to build the workspace, it showed some errors as follows. Anyone know how to solve the problem? Is it the version problem with my Gazebo? Thanks in advance.

> /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc: In member function ‘virtual void gazebo::UnitreeDrawForcePlugin::Load(gazebo::rendering::VisualPtr, sdf::v9::ElementPtr)’:
> /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:47:95: error: invalid use of incomplete type ‘class gazebo::common::Color’
>    47 |             this->line->AddPoint(ignition::math::Vector3d(0, 0, 0), common::Color(0, 1, 0, 1.0));
>       |                                                                                               ^
> In file included from /usr/include/gazebo-11/gazebo/common/Console.hh:30,
>                  from /usr/include/gazebo-11/gazebo/common/Events.hh:23,
>                  from /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:7:
> /usr/include/gazebo-11/gazebo/common/CommonTypes.hh:76:11: note: forward declaration of ‘class gazebo::common::Color’
>    76 |     class Color;
>       |           ^~~~~
> /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:48:95: error: invalid use of incomplete type ‘class gazebo::common::Color’
>    48 |             this->line->AddPoint(ignition::math::Vector3d(1, 1, 1), common::Color(0, 1, 0, 1.0));
>       |                                                                                               ^
> In file included from /usr/include/gazebo-11/gazebo/common/Console.hh:30,
>                  from /usr/include/gazebo-11/gazebo/common/Events.hh:23,
>                  from /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:7:
> /usr/include/gazebo-11/gazebo/common/CommonTypes.hh:76:11: note: forward declaration of ‘class gazebo::common::Color’
>    76 |     class Color;
>       |           ^~~~~
> /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:49:37: error: cannot convert ‘const char [14]’ to ‘const MaterialPtr&’ {aka ‘const Ogre::SharedPtr<Ogre::Material>&’}
>    49 |             this->line->setMaterial("Gazebo/Purple");
>       |                                     ^~~~~~~~~~~~~~~
>       |                                     |
>       |                                     const char [14]
> In file included from /usr/local/include/OGRE/Ogre.h:114,
>                  from /usr/include/gazebo-11/gazebo/rendering/ogre_gazebo.h:26,
>                  from /usr/include/gazebo-11/gazebo/rendering/Conversions.hh:25,
>                  from /usr/include/gazebo-11/gazebo/rendering/DynamicLines.hh:26,
>                  from /home/szwang/catkin_ws/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:12:
> /usr/local/include/OGRE/OgreSimpleRenderable.h:76:53: note:   initializing argument 1 of ‘virtual void Ogre::SimpleRenderable::setMaterial(const MaterialPtr&)’
>    76 |         virtual void setMaterial(const MaterialPtr& mat);
>       |                                  ~~~~~~~~~~~~~~~~~~~^~~
> make[2]: *** [unitree_ros/unitree_gazebo/CMakeFiles/unitreeDrawForcePlugin.dir/build.make:76: unitree_ros/unitree_gazebo/CMakeFiles/unitreeDrawForcePlugin.dir/plugin/draw_force_plugin.cc.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:3504: unitree_ros/unitree_gazebo/CMakeFiles/unitreeDrawForcePlugin.dir/all] Error 2
> make: *** [Makefile:146: all] Error 2
> Invoking "make -j4 -l4" failed

Many thanks.

Zhaiweiwei0, 2021-07-30:
It's most likely a version problem，we didn't have this problem with Ubuntu18.04.

ch1102256774, 2021-08-03:
The ubuntu and Gazebo environment I use are the same as yours, and I have encountered the same problems. It can be seen from the error message that the error is the following two lines of code in "unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc". 
```
 47     this->line->AddPoint(ignition::math::Vector3d(0, 0, 0), common::Color(0, 1, 0, 1.0)); 

 48     this->line->AddPoint(ignition::math::Vector3d(1, 1, 1), common::Color(0, 1, 0, 1.0));
```

I changed the 47th and 48th lines in the file to the following, and the compilation was successful, maybe you can try:
  ``` 
47     this->line->AddPoint(ignition::math::Vector3d(0, 0, 0), ignition::math::Color(0, 1, 0, 1.0));
            
48     this->line->AddPoint(ignition::math::Vector3d(1, 1, 1), ignition::math::Color(0, 1, 0, 1.0));
```
Hope my method can give you help.

edward9503, 2021-08-05:
> I changed the 47th and 48th lines in the file to the following, and the compilation was successful, maybe you can try:

> The ubuntu and Gazebo environment I use are the same as yours, and I have encountered the same problems. It can be seen from the error message that the error is the following two lines of code in [...]

edward9503, 2021-08-05:
> The ubuntu and Gazebo environment I use are the same as yours, and I have encountered the same problems. It can be seen from the error message that the error is the following two lines of code in [...]

Hi, thanks for your suggestions. They are really helpful!

Besides line 47 and 48,  there is also one error in line 49, which is:

> /home/szwang/leggedRob/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc: In member function ‘virtual void gazebo::UnitreeDrawForcePlugin::Load(gazebo::rendering::VisualPtr, sdf::v9::ElementPtr)’:
> /home/szwang/leggedRob/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:49:37: error: cannot convert ‘const char [14]’ to ‘const MaterialPtr&’ {aka ‘const Ogre::SharedPtr<Ogre::Material>&’}
>    49 |             this->line->setMaterial("Gazebo/Purple");
>       |                                     ^~~~~~~~~~~~~~~
>       |                                     |
>       |                                     const char [14]
> In file included from /usr/local/include/OGRE/Ogre.h:114,
>                  from /usr/include/gazebo-11/gazebo/rendering/ogre_gazebo.h:26,
>                  from /usr/include/gazebo-11/gazebo/rendering/Conversions.hh:25,
>                  from /usr/include/gazebo-11/gazebo/rendering/DynamicLines.hh:26,
>                  from /home/szwang/leggedRob/src/unitree_ros/unitree_gazebo/plugin/draw_force_plugin.cc:12:
> /usr/local/include/OGRE/OgreSimpleRenderable.h:76:53: note:   initializing argument 1 of ‘virtual void Ogre::SimpleRenderable::setMaterial(const MaterialPtr&)’
>    76 |         virtual void setMaterial(const MaterialPtr& mat);
>       |                                  ~~~~~~~~~~~~~~~~~~~^~~
> make[2]: *** [CMakeFiles/unitreeDrawForcePlugin.dir/build.make:76: CMakeFiles/unitreeDrawForcePlugin.dir/plugin/draw_force_plugin.cc.o] Error 1
> make[1]: *** [CMakeFiles/Makefile2:2323: CMakeFiles/unitreeDrawForcePlugin.dir/all] Error 2
> make: *** [Makefile:146: all] Error 2

Have you also encountered this problem? And how did you solve it? Many thanks!

ch1102256774, 2021-08-07:
Sorry, i didn't meet this problem. But it seems like a Parameter type error. 
I recommend you first Define a variable of type "MaterialPtr" .Then one of the attributes of this variable need to be set to  "Gazebo/Purple". Finally replace the "Gazebo/Purple" with the variable in line 49.
This is because a const char don't have a pointer, you should change it to the certain object which the function used in line 49 is expected.

staycoolish, 2022-06-22:
Thank you very much @ch1102256774 . I encountered the same problem in Noetic with Gazebo 11, and your solution worked for me.