unitreerobotics/unitree_ros #20: Problem with LCM server

AlaricLcrs, 2021-09-22:
Hello ! I have an a1 robot and have successfully setup my workspace. But form times to times, I get the following error when I run the robot in high level mode for example (**roslaunch mbs_unitree_ros high_level_mode.launch**) : 

> [node_lcm_server-2] process has died [pid 6969, exit code 255, cmd /home/amin/catkin_ws/devel/lib/unitree_legged_real/lcm_server_3_2 a1 highlevel __name:=node_lcm_server __log:=/root/.ros/log/37556312-1b72-11ec-a5c3-000c296e3085/node_lcm_server-2.log].
> log file: /root/.ros/log/37556312-1b72-11ec-a5c3-000c296e3085/node_lcm_server-2*.log

It occurs very randomly but often enough to become a problem. It usually works for a few minutes if I'm lucky and then I get the error, it doesn't occur right after I run the command. And I've tried to check the log file (_/root/.ros/log/37556312-1b72-11ec-a5c3-000c296e3085/node_lcm_server-2*.log_) but it doesn't exist though the folder does (and contains other log files). Any idea on how to solve it ?

PS : Here's the whole return message that I get.
[log.txt](https://github.com/unitreerobotics/unitree_ros/files/7208700/log.txt)

yiyuc, 2021-09-22:
Hi,

You need to kill the gzserver process cuz sometimes gazebo doesn’t kill the
process itself when exiting the simulator.

Best,
Yiyu

AlaricLcrs, 2021-09-23:
Thank you for this answer ! However, I'm not running gazebo, I'm using the real life robot so naturally i can't find a gzserver process to kill :') Is there another process that I should kill that'd be related to the use of the real robot ?

yiyuc, 2021-09-23:
Oh, I see, sorry I didn’t answer your question. This is the LCM timeout
error. It’s related to the lcm server when sending and receiving messages.
Are you running a real-time Ubuntu system? When you run code with robot, I
would suggest you using the sdk instead of ROS because ROS is not running
in real time and may cause the LCM timeout error. If you patch your OS with
the RT kernel, it should solve the issue but still, you may run into this
error some times.

Best,
Yiyu

AlaricLcrs, 2021-09-23:
Thanks for your quick reply ! 

I'm running Ubuntu on a virtual machine on a MacBook 😅
How would I use the sdk without ROS ? I thought I needed to use both of them to get the robot running 

Best,
Alaric

yiyuc, 2021-09-23:
I suggest you run it on a computer with Ubuntu. There is a separate sdk
repo you can find in the Unitree github and there are instructions there

AlaricLcrs, 2021-09-23:
Yes, the SDK is already installed on my workspace and I can start the lcm server but to communicate with the robot I still have to use ROS, no ? I didn't find instructions related to that there

yiyuc, 2021-09-24:
Check this repo, https://github.com/unitreerobotics/unitree_legged_sdk although the Aliengo robot uses version 3_2, but the build instruction is the same. You can run the example files by sudo ./example_

AlaricLcrs, 2021-09-29:
Okay thanks, it works now ! And last question I have : I've seen videos of the A1 dancing or doing backflips, is there an existing code that I could get or should I write the code myself ? 

Best,
Alaric

closed 2022-02-24 by AlaricLcrs