unitreerobotics/unitree_ros #21: Before runing unitree_servo, the robot shifts in Gazebo

edward9503, 2021-10-05:
Hi there,

when I run the gazebo simulation launch file and before I run the servo.cpp to control the robot to stand up, the robot always shifts towards some random directios. I was wondering is it a bug? And how to solve this problem? Because I want to fix the base postion and orientation, as well as the feet position after the robot is standing up. Many thanks!

ZKBian, 2021-10-08:
Hi edward,

This problem is related to Gazebo's surface contact simulation. Once the robot standing up, there are only point contact, then the robot will not shift. If you want to fix this problem, I guess you can remove the collision box of robot main body, and add 4 point contact on the bottom of main body

closed 2021-10-08 by ZKBian

edward9503, 2021-10-11:
> Hi edward,
>
> This problem is related to Gazebo's surface contact simulation. Once the robot standing up, there are only point contact, then the robot will not shift. If you want to fix this problem, [...]

Hi 

thanks for the reply. Because I am not familiar with the urdf or xacro, I was wondering can I add 4 point contacts for the body while keeping the collision box of the body at the same time?

Best