unitreerobotics/unitree_ros #35: resource not found: a1_gazebo

smileyenot983, 2022-02-17:
When i am running 
`roslaunch` unitree_gazebo normal.launch rname:=a1`

It gives me the  following errors:
xacro: in-order processing became default in ROS Melodic. You can drop the option.
resource not found: a1_gazebo
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/ramil/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
when processing file: /home/ramil/catkin_ws/src/unitree_ros/robots/a1_description/xacro/robot.xacro
RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '--inorder', '/home/ramil/catkin_ws/src/unitree_ros/robots/a1_description/xacro/robot.xacro', 'DEBUG:=false']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro --inorder '$(arg dollar)$(arg robot_path)/xacro/robot.xacro'             DEBUG:=$(arg user_debug)" name="robot_description"/>
The traceback for the exception was written to the log file

However, when i am running gazebo for laikago by writing `roslaunch` unitree_gazebo normal.launch rname:=laikago` everything works well

shirellm, 2022-02-21:
I got the same problem for both a1 and go1

smileyenot983, 2022-02-21:
After investing some time into this problem it turned out that the problem is in xacro. The problem is that xacro parses all comments in .xacro files. Thus, if there is some path which does not exist xacro will return an error, even though it is commented. You can read about it here https://github.com/ros/xacro/pull/300 .Should be fixed soon. Issue may be closed.

smileyenot983, 2022-02-21:
As a temporary fix you can just delete the comment in robot description xacro. In case of a1 it is 
``

shirellm, 2022-02-21:
> 
worked thank you!