unitreerobotics/unitree_ros #36: [urdf_spawner-4] process has died

zxydcl, 2022-03-03:
After I deleted the comment and successfully opened the models in the Rviz by the guidance in Issue #35 , I found I have problem in running the simulation by the command  `roslaunch unitree_gazebo normal.launch ` in the terminal.
It gave me the following errors:
```
[urdf_spawner-4] process has died [pid 17950, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -urdf -z 0.6 -model laikago_gazebo -param robot_description -unpause __name:=urdf_spawner __log:=/home/xinyu/.ros/log/00c0cee8-9aef-11ec-976c-103d1c3e9504/urdf_spawner-4.log].
log file: /home/xinyu/.ros/log/00c0cee8-9aef-11ec-976c-103d1c3e9504/urdf_spawner-4*.log
```
The complete terminal output was as follows:
```
... logging to /home/xinyu/.ros/log/00c0cee8-9aef-11ec-976c-103d1c3e9504/roslaunch-Office-17909.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://Office:37841/

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /laikago_gazebo/FL_calf_controller/joint: FL_calf_joint
 * /laikago_gazebo/FL_calf_controller/pid/d: 8.0
 * /laikago_gazebo/FL_calf_controller/pid/i: 0.0
 * /laikago_gazebo/FL_calf_controller/pid/p: 300.0
 * /laikago_gazebo/FL_calf_controller/type: unitree_legged_co...
 * /laikago_gazebo/FL_hip_controller/joint: FL_hip_joint
 * /laikago_gazebo/FL_hip_controller/pid/d: 5.0
 * /laikago_gazebo/FL_hip_controller/pid/i: 0.0
 * /laikago_gazebo/FL_hip_controller/pid/p: 100.0
 * /laikago_gazebo/FL_hip_controller/type: unitree_legged_co...
 * /laikago_gazebo/FL_thigh_controller/joint: FL_thigh_joint
 * /laikago_gazebo/FL_thigh_controller/pid/d: 8.0
 * /laikago_gazebo/FL_thigh_controller/pid/i: 0.0
 * /laikago_gazebo/FL_thigh_controller/pid/p: 300.0
 * /laikago_gazebo/FL_thigh_controller/type: unitree_legged_co...
 * /laikago_gazebo/FR_calf_controller/joint: FR_calf_joint
 * /laikago_gazebo/FR_calf_controller/pid/d: 8.0
 * /laikago_gazebo/FR_calf_controller/pid/i: 0.0
 * /laikago_gazebo/FR_calf_controller/pid/p: 300.0
 * /laikago_gazebo/FR_calf_controller/type: unitree_legged_co...
 * /laikago_gazebo/FR_hip_controller/joint: FR_hip_joint
 * /laikago_gazebo/FR_hip_controller/pid/d: 5.0
 * /laikago_gazebo/FR_hip_controller/pid/i: 0.0
 * /laikago_gazebo/FR_hip_controller/pid/p: 100.0
 * /laikago_gazebo/FR_hip_controller/type: unitree_legged_co...
 * /laikago_gazebo/FR_thigh_controller/joint: FR_thigh_joint
 * /laikago_gazebo/FR_thigh_controller/pid/d: 8.0
 * /laikago_gazebo/FR_thigh_controller/pid/i: 0.0
 * /laikago_gazebo/FR_thigh_controller/pid/p: 300.0
 * /laikago_gazebo/FR_thigh_controller/type: unitree_legged_co...
 * /laikago_gazebo/RL_calf_controller/joint: RL_calf_joint
 * /laikago_gazebo/RL_calf_controller/pid/d: 8.0
 * /laikago_gazebo/RL_calf_controller/pid/i: 0.0
 * /laikago_gazebo/RL_calf_controller/pid/p: 300.0
 * /laikago_gazebo/RL_calf_controller/type: unitree_legged_co...
 * /laikago_gazebo/RL_hip_controller/joint: RL_hip_joint
 * /laikago_gazebo/RL_hip_controller/pid/d: 5.0
 * /laikago_gazebo/RL_hip_controller/pid/i: 0.0
 * /laikago_gazebo/RL_hip_controller/pid/p: 100.0
 * /laikago_gazebo/RL_hip_controller/type: unitree_legged_co...
 * /laikago_gazebo/RL_thigh_controller/joint: RL_thigh_joint
 * /laikago_gazebo/RL_thigh_controller/pid/d: 8.0
 * /laikago_gazebo/RL_thigh_controller/pid/i: 0.0
 * /laikago_gazebo/RL_thigh_controller/pid/p: 300.0
 * /laikago_gazebo/RL_thigh_controller/type: unitree_legged_co...
 * /laikago_gazebo/RR_calf_controller/joint: RR_calf_joint
 * /laikago_gazebo/RR_calf_controller/pid/d: 8.0
 * /laikago_gazebo/RR_calf_controller/pid/i: 0.0
 * /laikago_gazebo/RR_calf_controller/pid/p: 300.0
[... 58 lines cut ...]
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 171, in get_manifest
    return self._load_manifest(name)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 215, in _load_manifest
    retval = self._manifests[name] = parse_manifest_file(self.get_path(name), self._manifest_name, rospack=self)
  File "/usr/lib/python2.7/dist-packages/rospkg/manifest.py", line 414, in parse_manifest_file
    _static_rosdep_view = init_rospack_interface()
  File "/usr/lib/python2.7/dist-packages/rosdep2/rospack.py", line 60, in init_rospack_interface
    lookup = _get_default_RosdepLookup(Options())
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in _get_default_RosdepLookup
    verbose=options.verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 609, in create_default
    sources = load_cached_sources_list(sources_cache_dir=sources_cache_dir, verbose=verbose)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 566, in load_cached_sources_list
    raise CachePermissionError('Failed to write cache file: ' + str(e))
rosdep2.core.CachePermissionError: Failed to write cache file: [Errno 13] Permission denied: '/home/xinyu/.ros/rosdep/sources.cache/index'
[urdf_spawner-4] process has died [pid 17950, exit code 1, cmd /opt/ros/melodic/lib/gazebo_ros/spawn_model -urdf -z 0.6 -model laikago_gazebo -param robot_description -unpause __name:=urdf_spawner __log:=/home/xinyu/.ros/log/00c0cee8-9aef-11ec-976c-103d1c3e9504/urdf_spawner-4.log].
log file: /home/xinyu/.ros/log/00c0cee8-9aef-11ec-976c-103d1c3e9504/urdf_spawner-4*.log
[ INFO] [1646311185.992605882]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1646311186.003600256]: Physics dynamic reconfigure ready.
[WARN] [1646311215.255998, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[laikago_gazebo/controller_spawner-5] process has finished cleanly
log file: /home/xinyu/.ros/log/00c0cee8-9aef-11ec-976c-103d1c3e9504/laikago_gazebo-controller_spawner-5*.log
```

I've successfully run the simulation before, it's weird to have a error like this.

Dong-No, 2022-07-15:
I'm not sure why but changing the rosdep permission solve this.
$ sudo rosdep fix-permissions

xyq1223, 2023-03-23:
Hello,you can check the runtime environment of the program.

JadisHee, 2024-01-04:
I've got the same problem, just change DISTRIB_ID in /etc/lsb-realse from "kylin" to "debian" will be fine

Ethereal1679, 2025-07-29:
> I've got the same problem, just change DISTRIB_ID in /etc/lsb-realse from "kylin" to "debian" will be fine

Sorry, but mine is "Ubuntu", Can I change it to "debian"?