unitreerobotics/unitree_ros #38: boost signals error while compiling

Segev-Cohen, 2022-03-13:
I cloned the package to src directory and tried to compile it but I have encountered a problem with the boost package.

`-- +++ processing catkin package: 'unitree_controller'
-- ==> add_subdirectory(unitree_ros/unitree_controller)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /usr/lib/cmake/Boost-1.78.0/BoostConfig.cmake:141 (find_package):
  Could not find a package configuration file provided by "boost_signals"
  (requested version 1.78.0) with any of the following names:

    boost_signalsConfig.cmake
    boost_signals-config.cmake

  Add the installation prefix of "boost_signals" to CMAKE_PREFIX_PATH or set
  "boost_signals_DIR" to a directory containing one of the above files.  If
  "boost_signals" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/lib/cmake/Boost-1.78.0/BoostConfig.cmake:262 (boost_find_component)
  /usr/share/cmake-3.10/Modules/FindBoost.cmake:242 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/gazebo/gazebo-config.cmake:159 (find_package)
  unitree_ros/unitree_controller/CMakeLists.txt:17 (find_package)
`

Have anyone got an idea what may cause this problem?

arschoon, 2022-06-27:
@Segev-Cohen How did you resolve the boost version issue? I am running into the same error

Panting007, 2024-06-03:
How did you solve this problem? I had the same problem.