unitreerobotics/unitree_ros #43: Problem with buid

JonathanMortes, 2022-04-09:
I've got the catkin folder setup and all dependencies installed but I keep getting this same set of errors: (maybe a tutorial to set everything up could be helpful)

Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_BmsCmd
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_BmsCmd
Scanning dependencies of target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_BmsState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_BmsState
Scanning dependencies of target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_nodejs
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_MotorState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_MotorState
Scanning dependencies of target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_nodejs
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_LowState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_LowState
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_LowCmd
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_LowCmd
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_HighState
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_HighState
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_Cartesian
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_Cartesian
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_MotorCmd
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_MotorCmd
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_IMU
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_IMU
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_HighCmd
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_HighCmd
Scanning dependencies of target _unitree_legged_msgs_generate_messages_check_deps_LED
[  0%] Built target _unitree_legged_msgs_generate_messages_check_deps_LED
Scanning dependencies of target unitree_legged_msgs_generate_messages_nodejs
[  1%] Generating Javascript code from unitree_legged_msgs/LowState.msg
[  2%] Generating Javascript code from unitree_legged_msgs/LowCmd.msg
[  4%] Generating Javascript code from unitree_legged_msgs/HighState.msg
[  5%] Generating Javascript code from unitree_legged_msgs/BmsCmd.msg
[  7%] Generating Javascript code from unitree_legged_msgs/MotorCmd.msg
[  8%] Generating Javascript code from unitree_legged_msgs/IMU.msg
[ 10%] Generating Javascript code from unitree_legged_msgs/Cartesian.msg
[ 11%] Generating Javascript code from unitree_legged_msgs/HighCmd.msg
[ 13%] Generating Javascript code from unitree_legged_msgs/LED.msg
[ 14%] Generating Javascript code from unitree_legged_msgs/BmsState.msg
[ 16%] Generating Javascript code from unitree_legged_msgs/MotorState.msg
[ 16%] Built target unitree_legged_msgs_generate_messages_nodejs
Scanning dependencies of target geometry_msgs_generate_messages_eus
[ 16%] Built target geometry_msgs_generate_messages_eus
Scanning dependencies of target sensor_msgs_generate_messages_eus
[ 16%] Built target sensor_msgs_generate_messages_eus
Scanning dependencies of target std_msgs_generate_messages_eus
[ 16%] Built target std_msgs_generate_messages_eus
Scanning dependencies of target unitree_legged_msgs_generate_messages_eus
[ 17%] Generating EusLisp code from unitree_legged_msgs/LowState.msg
[ 19%] Generating EusLisp code from unitree_legged_msgs/LowCmd.msg
[ 20%] Generating EusLisp code from unitree_legged_msgs/HighState.msg
[ 22%] Generating EusLisp code from unitree_legged_msgs/BmsCmd.msg
[ 23%] Generating EusLisp code from unitree_legged_msgs/MotorCmd.msg
[ 25%] Generating EusLisp code from unitree_legged_msgs/IMU.msg
[ 26%] Generating EusLisp code from unitree_legged_msgs/Cartesian.msg
[ 28%] Generating EusLisp code from unitree_legged_msgs/HighCmd.msg
[ 29%] Generating EusLisp code from unitree_legged_msgs/LED.msg
[ 31%] Generating EusLisp code from unitree_legged_msgs/BmsState.msg
[ 32%] Generating EusLisp code from unitree_legged_msgs/MotorState.msg
[ 34%] Generating EusLisp manifest code for unitree_legged_msgs
[ 34%] Built target unitree_legged_msgs_generate_messages_eus
Scanning dependencies of target geometry_msgs_generate_messages_py
[ 34%] Built target geometry_msgs_generate_messages_py
Scanning dependencies of target std_msgs_generate_messages_py
[ 34%] Built target std_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_py
[ 34%] Built target sensor_msgs_generate_messages_py
Scanning dependencies of target unitree_legged_msgs_generate_messages_py
[... 134 lines cut ...]
/home/jonathan/catkin_ws/src/unitree_legged_real/include/convert.h:198:18: error: no match for ‘operator=’ (operand types are ‘std::array<unsigned int, 2>’ and ‘unitree_legged_msgs::HighCmd_<std::allocator<void> >::_SN_type {aka unsigned int}’)
     lcm.SN = ros.SN;
                  ^~
In file included from /usr/include/c++/7/tuple:39:0,
                 from /usr/include/c++/7/bits/unique_ptr.h:37,
                 from /usr/include/c++/7/bits/locale_conv.h:41,
                 from /usr/include/c++/7/locale:43,
                 from /usr/include/c++/7/iomanip:43,
                 from /usr/include/boost/math/policies/error_handling.hpp:12,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/melodic/include/ros/time.h:58,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/jonathan/catkin_ws/src/unitree_legged_real/src/exe/position_mode.cpp:6:
/usr/include/c++/7/array:94:12: note: candidate: std::array<unsigned int, 2>& std::array<unsigned int, 2>::operator=(const std::array<unsigned int, 2>&)
     struct array
            ^~~~~
/usr/include/c++/7/array:94:12: note:   no known conversion for argument 1 from ‘unitree_legged_msgs::HighCmd_<std::allocator<void> >::_SN_type {aka unsigned int}’ to ‘const std::array<unsigned int, 2>&’
/usr/include/c++/7/array:94:12: note: candidate: std::array<unsigned int, 2>& std::array<unsigned int, 2>::operator=(std::array<unsigned int, 2>&&)
/usr/include/c++/7/array:94:12: note:   no known conversion for argument 1 from ‘unitree_legged_msgs::HighCmd_<std::allocator<void> >::_SN_type {aka unsigned int}’ to ‘std::array<unsigned int, 2>&&’
unitree_legged_real/CMakeFiles/position_lcm.dir/build.make:62: recipe for target 'unitree_legged_real/CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o' failed
make[2]: *** [unitree_legged_real/CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o] Error 1
CMakeFiles/Makefile2:1886: recipe for target 'unitree_legged_real/CMakeFiles/position_lcm.dir/all' failed
make[1]: *** [unitree_legged_real/CMakeFiles/position_lcm.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

Affonso-Gui, 2022-04-10:
Maybe https://github.com/unitreerobotics/unitree_ros_to_real/pull/25 fixes your problem.
More specifically: [unitreerobotics/unitree_ros_to_real@`49acae3` (#25)](https://github.com/unitreerobotics/unitree_ros_to_real/pull/25/commits/49acae319f7236d47751b9cb1a8184ab3c29809e)

andrewhstovell, 2022-09-07:
Checkout my responses on #53 these may help