unitreerobotics/unitree_ros #44: unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory

erencanbulut, 2022-04-22:
I copied and pasted "mbs_simulation" file to my new catkin_ws folder. Then, I got this problem when I tried to make "catkin build"

**The problem:** unitree_legged_sdk/unitree_legged_sdk.h: No such file or directory

It seems like: (https://user-images.githubusercontent.com/54208292/164712498-fdbedc1d-cd85-48c1-a5c4-50c148560723.png)

HDElectronics, 2022-05-28:
Hey, I got the same issue, any fix available ?

Infas12, 2022-06-08:
Have you installed [unitree_legged_sdk](https://github.com/unitreerobotics/unitree_legged_sdk) ?

erencanbulut, 2022-06-10:
> Hey, I got the same issue, any fix available ?

Unfortunately not.

andrewhstovell, 2022-10-26:
Hi @erencanbulut & @HDElectronics,

_Disclaimer: I've had and resolved this issue. Though the cause of the problem for me may differ from yours depending on versions of the packages used. For me, this issue occurred when using unitree_ros_to_real v3.4.0, along with unitree_legged_sdk v3.4.2_

This is not a **unitree_ros** issue, but rather a **unitree_ros_to_real** issue (Its also not really an issue as its mentioned in the ReadME), to resolve this, edit the paths to the unitree_legged_sdk in the CMakeLists.txt file located in src/unitree_ros_to_real/unitree_legged_real/ to the correct paths for your workspace.

Originally, 
```
# line 21 onward
include_directories(/home/$ENV{USER}/Robot_SDK/unitree_legged_sdk/include)
link_directories(/home/$ENV{USER}/Robot_SDK/unitree_legged_sdk/lib)
string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_amd64.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} lcm)

add_executable(lcm_server /home/$ENV{USER}/Robot_SDK/unitree_legged_sdk/examples/lcm_server.cpp)
target_link_libraries(lcm_server ${EXTRA_LIBS} ${catkin_LIBRARIES})
add_dependencies(lcm_server ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
```

After editing, (In this example my packages are stored in /home/andy/catkin_ws/src
```
# line 21 onward
include_directories(/home/$ENV{USER}/catkin_ws/src/unitree_legged_sdk/include)
link_directories(/home/$ENV{USER}/catkin_ws/src/unitree_legged_sdk/lib)
string(CONCAT LEGGED_SDK_NAME libunitree_legged_sdk_amd64.so)
set(EXTRA_LIBS ${LEGGED_SDK_NAME} lcm)

add_executable(lcm_server /home/$ENV{USER}/catkin_ws/src/unitree_legged_sdk/examples/lcm_server.cpp)
target_link_libraries(lcm_server ${EXTRA_LIBS} ${catkin_LIBRARIES})
add_dependencies(lcm_server ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
```

Finally, make sure that you are running catkin_make as the correct user, for example, if i ran the above as  `root` the linker would search in `/home/root/catkin_ws/src/`

Hope this helps!

<hr/>
From [unitree_ros_to_real / v3.4.0](https://github.com/unitreerobotics/unitree_ros_to_real/tree/v3.4.0) ReadME:

> Before compiling this package, users have to modify the path to unitree_legged_sdk under the CMakeLists.txt of unitree_legged_real. Just search the "/home/bian/Robot_SDK/unitree_legged_sdk" and change it to your own path. If you are going to compile this package on a computer which is not AMD64 platform, then you have to search "libunitree_legged_sdk.so" under the CMakeLists.txt and change it to the .so file name which is suitable to your computer.
>