unitreerobotics/unitree_ros #47: Control the Go1 robot with the topic and rosmaster

fLattes, 2022-06-09:
Hello, 
I tried to use directly the topic to receive information about camera... and to send information in topics. 
For that, we use the network connection with rosmaster, and in WiFi. We can see all the topics of the robot but the problem is that we didn't receive any information with the command "rostopic echo /topic". 
Can you help me, please.

andrewhstovell, 2022-10-26:
For the cameras i think you want to look at:
[unitree_camera_sdk](https://github.com/unitreerobotics/UnitreecameraSDK)

xyq1223, 2023-03-23:
Hello,firstly, please make sure that the communication is working, and secondly, we are sorry that we do not provide ROS routines in our camera.