unitreerobotics/unitree_ros #49: CMake error (unitree_legged_msgs)

CarlosBambinos, 2022-07-20:
Hi everyone,
I'm a beginner in ROS and I looked on several forums but didn't get the solution for this specific problem of package if i understand well.
I cloned the Unitree_ros Git and followed the Tutorial under but when i launch the "catkin_make" I have this error:

CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "unitree_legged_msgs" with any of the following names:

    unitree_legged_msgsConfig.cmake
    unitree_legged_msgs-config.cmake

  Add the installation prefix of "unitree_legged_msgs" to CMAKE_PREFIX_PATH
  or set "unitree_legged_msgs_DIR" to a directory containing one of the above
  files.  If "unitree_legged_msgs" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  unitree_ros/unitree_legged_control/CMakeLists.txt:4 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/odysse/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/odysse/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:320: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Have you ever get it and if yes how can I solve it please?

andrewhstovell, 2022-09-07:
Hi @CarlosBambinos I posted a pdf on #53 with detailed instructions. this may help

mrasamu, 2022-12-06:
https://github.com/unitreerobotics/unitree_ros_to_real/tree/master/unitree_legged_msgs

xyq1223, 2023-03-22:
Hello,you can download the latest ros_to_real package.
https://github.com/unitreerobotics/unitree_ros_to_real/blob/master/unitree_legged_msgs/msg/LowState.msg