unitreerobotics/unitree_ros #50: Build error

Ruslan2288, 2022-08-07:
Errors     << unitree_controller:make /workspace/logs/unitree_controller/build.make.005.log                                           
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp: In member function 'void multiThread::FRfootCallback(const WrenchStamped&)':
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:170:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[0].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:171:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[0].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:172:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[0].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp: In member function 'void multiThread::FLfootCallback(const WrenchStamped&)':
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:178:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[1].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:179:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[1].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:180:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[1].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp: In member function 'void multiThread::RRfootCallback(const WrenchStamped&)':
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:186:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[2].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:187:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[2].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:188:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[2].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp: In member function 'void multiThread::RLfootCallback(const WrenchStamped&)':
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:194:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[3].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:195:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[3].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/workspace/src/unitree_ros/unitree_controller/src/servo.cpp:196:18: error: 'unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}' has no member named 'eeForce'; did you mean 'footForce'?
         lowState.eeForce[3].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
make[2]: *** [CMakeFiles/unitree_servo.dir/src/servo.cpp.o] Error 1
make[1]: *** [CMakeFiles/unitree_servo.dir/all] Error 2
make: *** [all] Error 2
cd /workspace/build/unitree_controller; catkin build --get-env unitree_controller | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -                                                           
..................................................................
Failed     << unitree_controller:make               [ Exited with code 2 ]                                                            
Failed    <<< unitree_controller                    [ 6.8 seconds ]
[build] Summary: 6 of 7 packages succeeded.                       
[build]   Ignored:   None.                                        
[build]   Warnings:  None.                                        
[build]   Abandoned: None.                                        
[build]   Failed:    1 packages failed.                           
[build] Runtime: 7.2 seconds total.

andrewhstovell, 2022-08-12:
Having the same issue.

I notice that 
unitree_ros_to_real/../unitree_legged_msgs::LowState does not, _and never has_ contained an 'eeForce' variable.
Meanwhile, 
unitree_ros/unitree_controller/src/servo.cpp has always referred to it in all versions. 

We must be missing something here..

andrewhstovell, 2022-08-12:
**Commit: 1ad059e38dcaba37cd3f4a6600727e8f789be658 on the 24th May** saw the Cartesian eeForce get removed from unitree_legged_msgs::LowState. However its references clearly weren't checked.

Is the eeForce needed or not?

andrewhstovell, 2022-09-07:
@Ruslan2288 check out my responses to #53