unitreerobotics/unitree_ros #51: Build error

jothyhariraja, 2022-08-10:
*********************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.15
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x64'

C:\Windows\System32>call C:\catkin_ws\devel\setup.bash
'C:\catkin_ws\devel\setup.bash' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>call C:\catkin_ws\devel\setup.bat

C:\Windows\System32>call C:\opt\ros\melodic\x64\setup.bt
'C:\opt\ros\melodic\x64\setup.bt' is not recognized as an internal or external command,
operable program or batch file.

C:\Windows\System32>call C:\opt\ros\melodic\x64\setup.bat

C:\Windows\System32>rosrun unitre_controller unitree_servo
[rospack] Error: package 'unitre_controller' not found

C:\Windows\System32>call C:\catkin_ws\devel\setup.bat

C:\Windows\System32>rosrun unitre_controller unitree_servo
[rospack] Error: package 'unitre_controller' not found

C:\Windows\System32>cd  C:\catkin_ws

C:\catkin_ws>catkin_make
Base path: C:\catkin_ws
Source space: C:\catkin_ws\src
Build space: C:\catkin_ws\build
Devel space: C:\catkin_ws\devel
Install space: C:\catkin_ws\install
####
#### Running command: "nmake cmake_check_build_system" in "C:\catkin_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.29.30145.0
Copyright (C) Microsoft Corporation.  All rights reserved.

####
#### Running command: "nmake" in "C:\catkin_ws\build"
####

Microsoft (R) Program Maintenance Utility Version 14.29.30145.0
Copyright (C) Microsoft Corporation.  All rights reserved.

[  2%] Built target Project__setup_util.py_exec_install_python
[  2%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target std_msgs_generate_messages_lisp
[  2%] Built target sensor_msgs_generate_messages_lisp
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_HighCmd
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_Cartesian
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_LowState
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_LED
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_MotorState
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_BmsCmd
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_IMU
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_HighState
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_BmsState
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_MotorCmd
[  2%] Built target _unitree_legged_msgs_generate_messages_check_deps_LowCmd
[ 15%] Built target unitree_legged_msgs_generate_messages_lisp
[ 15%] Built target geometry_msgs_generate_messages_cpp
[ 15%] Built target sensor_msgs_generate_messages_cpp
[ 15%] Built target std_msgs_generate_messages_cpp
[ 28%] Built target unitree_legged_msgs_generate_messages_cpp
[ 28%] Built target geometry_msgs_generate_messages_py
[ 28%] Built target sensor_msgs_generate_messages_py
[ 28%] Built target std_msgs_generate_messages_py
[ 42%] Built target unitree_legged_msgs_generate_messages_py
[ 42%] Built target std_msgs_generate_messages_eus
[ 42%] Built target geometry_msgs_generate_messages_eus
[ 42%] Built target sensor_msgs_generate_messages_eus
[ 56%] Built target unitree_legged_msgs_generate_messages_eus
[ 56%] Built target geometry_msgs_generate_messages_nodejs
[ 56%] Built target sensor_msgs_generate_messages_nodejs
[ 56%] Built target std_msgs_generate_messages_nodejs
[ 69%] Built target unitree_legged_msgs_generate_messages_nodejs
[ 69%] Built target unitree_legged_msgs_generate_messages
[ 69%] Built target unitree_legged_msgs_gencpp
[ 70%] Linking CXX shared library C:\catkin_ws\devel\bin\unitree_legged_control.dll
LINK Pass 1: command "C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.301\bin\Hostx64\x64\link.exe /nologo @CMakeFiles\unitree_legged_control.dir\objects1.rsp /out:C:\catkin_ws\devel\bin\unitree_legged_control.dll /implib:C:\catkin_ws\devel\lib\unitree_legged_control.lib /pdb:C:\catkin_ws\devel\bin\unitree_legged_control.pdb /dll /version:0.0 /machine:x64 /debug /INCREMENTAL /DEF:CMakeFiles\unitree_legged_control.dir\exports.def -LIBPATH:C:\catkin_ws\src\unitree_ros\unitree_legged_control\ -LIBPATH:C:\catkin_ws\src\unitree_ros\unitree_legged_control\lib C:\opt\ros\melodic\x64\lib\class_loader.lib C:\opt\ros\melodic\x64\lib\PocoFoundation.lib C:\opt\ros\melodic\x64\lib\roslib.lib C:\opt\ros\melodic\x64\lib\rospack.lib C:\opt\ros\melodic\x64\libs\python27.lib C:\opt\ros\melodic\x64\lib\boost_program_options-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\tinyxml2.lib C:\opt\ros\melodic\x64\lib\realtime_tools.lib C:\opt\ros\melodic\x64\lib\roscpp.lib C:\opt\ros\melodic\x64\lib\boost_filesystem-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\rosconsole.lib C:\opt\ros\melodic\x64\lib\rosconsole_log4cxx.lib C:\opt\ros\melodic\x64\lib\rosconsole_backend_interface.lib C:\opt\ros\melodic\x64\lib\log4cxx.lib C:\opt\ros\melodic\x64\lib\boost_regex-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\xmlrpcpp.lib C:\opt\ros\melodic\x64\lib\roscpp_serialization.lib C:\opt\ros\melodic\x64\lib\rostime.lib C:\opt\ros\melodic\x64\lib\cpp_common.lib C:\opt\ros\melodic\x64\lib\boost_system-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\boost_thread-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\boost_chrono-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\boost_date_time-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\boost_atomic-vc141-mt-x64-1_66.lib C:\opt\ros\melodic\x64\lib\console_bridge.lib unitree_joint_control_tool.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:CMakeFiles\unitree_legged_control.dir/intermediate.manifest CMakeFiles\unitree_legged_control.dir/manifest.res" failed (exit code 1104) with the following output:
LINK : fatal error LNK1104: cannot open file 'unitree_joint_control_tool.lib'
NMAKE : fatal error U1077: 'C:\opt\ros\melodic\x64\lib\site-packages\cmake\data\bin\cmake.exe' : return code '0xffffffff'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64\nmake.exe"' : return code '0x2'
Stop.
Invoking "nmake" failed

jothyhariraja, 2022-08-10:
how to resolve this issue , please reply.

yiyuc, 2022-09-13:
Use ubuntu system instead of windows/WSL