unitreerobotics/unitree_ros #53: Problem with building unitree_controller package

wongearth, 2022-09-04:
my system 
ubuntu 18.04
unitree_legged_sdk: 3.8.0
cmake 3.10.2
lcm 1.4.0
g++ 8.4.0

I have problem with building the unitree_controller package. Can not get pass these error. Please.

Errors     << unitree_controller:make /home/wong/catkin_ws/logs/unitree_controller/build.make.011.log
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp: In member function ‘void multiThread::FRfootCallback(const WrenchStamped&)’:
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:170:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[0].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:171:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[0].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:172:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[0].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp: In member function ‘void multiThread::FLfootCallback(const WrenchStamped&)’:
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:178:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[1].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:179:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[1].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:180:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[1].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp: In member function ‘void multiThread::RRfootCallback(const WrenchStamped&)’:
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:186:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[2].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:187:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[2].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:188:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[2].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp: In member function ‘void multiThread::RLfootCallback(const WrenchStamped&)’:
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:194:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[3].x = msg.wrench.force.x;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:195:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[3].y = msg.wrench.force.y;
                  ^~~~~~~
                  footForce
/home/wong/catkin_ws/src/unitree_ros/unitree_controller/src/servo.cpp:196:18: error: ‘unitree_legged_msgs::LowState’ {aka ‘struct unitree_legged_msgs::LowState_<std::allocator<void> >’} has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[3].z = msg.wrench.force.z;
                  ^~~~~~~
                  footForce
make[2]: *** [CMakeFiles/unitree_servo.dir/src/servo.cpp.o] Error 1
make[1]: *** [CMakeFiles/unitree_servo.dir/all] Error 2
make: *** [all] Error 2
cd /home/wong/catkin_ws/build/unitree_controller; catkin build --get-env unitree_controller | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << unitree_controller:make            [ Exited with code 2 ]        
Failed    <<< unitree_controller                 [ 7.4 seconds ]

DrStoop, 2022-09-05:
Hi, I am having the same problem which is related to the change of the [`LowState.msg`](https://github.com/unitreerobotics/unitree_ros_to_real/blob/master/unitree_legged_msgs/msg/LowState.msg) in the [`unitreerobotics/unitree_ros_to_real`](https://github.com/unitreerobotics/unitree_ros_to_real)-repository. `eeForce` was removed from `LowState.msg` in the  @ZKBian's [commit](https://github.com/unitreerobotics/unitree_ros_to_real/commit/f8bbd5b1de17bad6cf9eb7aa8105e4a95fc52e85) and it was even [pointed out](https://github.com/unitreerobotics/unitree_ros_to_real/commit/1ad059e38dcaba37cd3f4a6600727e8f789be658#r80995642) by @andrewhstovell that `eeForce` is still used by [`servo.cpp`](https://github.com/unitreerobotics/unitree_ros/blob/92a36c7cf2fe7781adedce30cd6a6ab1456ef56c/unitree_controller/src/servo.cpp#L170). @ZKBian or @andrewhstovell, do you know how to resolve this issue/bug?

Thanks a lot in advance!

andrewhstovell, 2022-09-06:
Hi @Dr. Stoop and @ZKBian, Yes i did figure this out.

Technically this isn't an issue, it comes down to the versions of the packages used.

Please use the following package releases:

unitree_legged_sdk v3.4.2
commit: b5d6db22d08de59b0d1da835b40559b073fce415

unitree_ros_to_real v3.4.0
commit: 3c0fe3c97d048464869074c90719997731333841

unitree_ros (Most recent at time)
commit: 2e8877329ccb3b5b24d71f04f8ee9df68b3372e9

I actually compiled a complete list of instructions for cloning, setting up and building on Ubuntu 18.04, you may find it useful :)

[ROS Build instructions for Ubuntu.pdf](https://github.com/unitreerobotics/unitree_ros/files/9492749/ROS.Build.instructions.for.Ubuntu.pdf)

wongearth, 2022-09-06:
@andrewhstovell Thank you. I could now built unitree_controller successfully. But there seems to be others errors building unitree_legged_real.  A long list of errors... some are below. It seems to relate to lcm. Anyone has this problem? I installed lcm 1.4.0. 

In file included from /home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/src/exe/velocity_mode.cpp:13:
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h: In function ‘unitree_legged_msgs::LowState ToRos(UNITREE_LEGGED_SDK::LowState&)’:
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:108:27: error: ‘struct UNITREE_LEGGED_SDK::LowState’ has no member named ‘commVersion’; did you mean ‘version’?
     ros.commVersion = lcm.commVersion;
                           ^~~~~~~~~~~
                           version
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:109:23: error: ‘struct UNITREE_LEGGED_SDK::LowState’ has no member named ‘robotID’
     ros.robotID = lcm.robotID;
                       ^~~~~~~
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:110:18: error: cannot convert ‘std::array<unsigned int, 2>’ to ‘unitree_legged_msgs::LowState_<std::allocator<void> >::_SN_type’ {aka ‘unsigned int’} in assignment
     ros.SN = lcm.SN;
                  ^~
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h: In function ‘UNITREE_LEGGED_SDK::LowCmd ToLcm(unitree_legged_msgs::LowCmd&, UNITREE_LEGGED_SDK::LowCmd)’:
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:134:9: error: ‘struct UNITREE_LEGGED_SDK::LowCmd’ has no member named ‘commVersion’; did you mean ‘version’?
     lcm.commVersion = ros.commVersion;
         ^~~~~~~~~~~
         version
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:135:9: error: ‘struct UNITREE_LEGGED_SDK::LowCmd’ has no member named ‘robotID’
     lcm.robotID = ros.robotID;
         ^~~~~~~
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:136:18: error: no match for ‘operator=’ (operand types are ‘std::array<unsigned int, 2>’ and ‘unitree_legged_msgs::LowCmd_<std::allocator<void> >::_SN_type’ {aka ‘unsigned int’})
     lcm.SN = ros.SN;
                  ^~
In file included from /usr/include/c++/8/tuple:39,
                 from /usr/include/c++/8/bits/unique_ptr.h:37,
                 from /usr/include/c++/8/bits/locale_conv.h:41,
                 from /usr/include/c++/8/locale:43,
                 from /usr/include/c++/8/iomanip:43,
                 from /usr/include/boost/math/policies/error_handling.hpp:12,
                 from /usr/include/boost/math/special_functions/round.hpp:14,
                 from /opt/ros/melodic/include/ros/time.h:58,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/src/exe/velocity_mode.cpp:6:
/usr/include/c++/8/array:94:12: note: candidate: ‘std::array<unsigned int, 2>& std::array<unsigned int, 2>::operator=(const std::array<unsigned int, 2>&)’
     struct array
            ^~~~~
/usr/include/c++/8/array:94:12: note:   no known conversion for argument 1 from ‘unitree_legged_msgs::LowCmd_<std::allocator<void> >::_SN_type’ {aka ‘unsigned int’} to ‘const std::array<unsigned int, 2>&’
/usr/include/c++/8/array:94:12: note: candidate: ‘std::array<unsigned int, 2>& std::array<unsigned int, 2>::operator=(std::array<unsigned int, 2>&&)’
/usr/include/c++/8/array:94:12: note:   no known conversion for argument 1 from ‘unitree_legged_msgs::LowCmd_<std::allocator<void> >::_SN_type’ {aka ‘unsigned int’} to ‘std::array<unsigned int, 2>&&’
In file included from /home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/src/exe/velocity_mode.cpp:13:
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h: In function ‘unitree_legged_msgs::HighState ToRos(UNITREE_LEGGED_SDK::HighState&)’:
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:154:27: error: ‘struct UNITREE_LEGGED_SDK::HighState’ has no member named ‘commVersion’; did you mean ‘version’?
     ros.commVersion = lcm.commVersion;
                           ^~~~~~~~~~~
                           version
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:155:23: error: ‘struct UNITREE_LEGGED_SDK::HighState’ has no member named ‘robotID’
     ros.robotID = lcm.robotID;
                       ^~~~~~~
/home/mclab/catkin_ws/src/unitree_ros_to_real/unitree_legged_real/include/convert.h:156:18: error: cannot convert ‘std::array<unsigned int, 2>’ to ‘unitree_legged_msgs::HighState_<std::allocator<void> >::_SN_type’ {aka ‘unsigned int’} in assignment
     ros.SN = lcm.SN;

DrStoop, 2022-09-07:
thanks a lot @andrewhstovell, works perfectly for me 👌

andrewhstovell, 2022-09-07:
"A long list of errors"

@wongearth I do remember having this issue. But i cannot remember exactly how i resolved it.

I think it was resolved by changing the executable path to the lcm_server in the CMakeLists file. (See section 7 of the pdf i posted)

snijders-tjm, 2022-09-23:
@andrewhstovell Thank you for your detailed instruction file. I have followed it step by step and it worked, at least that is what I wonder. 

In the final step, after "roslaunch unitree_gazebo normal.launch rname:=go1 wname:=stairs", the simulation indeed starts and I see the go1 robot in the stairs environment. However, over time it shifts and falls of the platform, is this normal? I would have expected it to stay in place in order to initiate, for instance, the stand controller as in the Readme example.

Is this normal behaviour or did I possibly make an error anyway?

Thanks in advance!

xyq1223, 2023-03-28:
Hello,this is caused by using the wrong version of unitree_legged_msgs. Please select unitree_legged_msgs from the unitree_ros_to_real (v3.2.1) package
https://github.com/unitreerobotics/unitree_ros_to_real/tree/v3.2.1
 and put it under ~/catkin_ws/src and recompile it.

xyq1223, 2023-03-28:
Or you can choose to add the following statement to the unitree_legged_msgs/msg/LowState.msg file and recompile.

Cartesian[4] eeForceRaw
Cartesian[4] eeForce