unitreerobotics/unitree_ros #54: Cannot find -lunitree_legged_sdk_amd64

JulienRineau, 2022-10-26:
Hi,
when I try to catkin_make I obtain this error:
```
[ 82%] Building CXX object unitree_ros_to_real/unitree_legged_real/CMakeFiles/walk_lcm.dir/src/exe/walk_mode.cpp.o
[ 83%] Building CXX object unitree_ros_to_real/unitree_legged_real/CMakeFiles/position_lcm.dir/src/exe/position_mode.cpp.o
/usr/bin/ld: skipping incompatible /home/parallels/capstone/capstone_ws/src/unitree_legged_sdk/lib/libunitree_legged_sdk_amd64.so when searching for -lunitree_legged_sdk_amd64
/usr/bin/ld: cannot find -lunitree_legged_sdk_amd64
collect2: error: ld returned 1 exit status
make[2]: *** [unitree_ros_to_real/unitree_legged_real/CMakeFiles/torque_lcm.dir/build.make:101: /home/parallels/capstone/capstone_ws/devel/lib/unitree_legged_real/torque_lcm] Error 1
make[1]: *** [CMakeFiles/Makefile2:2697: unitree_ros_to_real/unitree_legged_real/CMakeFiles/torque_lcm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 85%] Linking CXX executable /home/parallels/capstone/capstone_ws/devel/lib/unitree_legged_real/walk_lcm
[ 86%] Linking CXX executable /home/parallels/capstone/capstone_ws/devel/lib/unitree_legged_real/position_lcm
/usr/bin/ld: skipping incompatible /home/parallels/capstone/capstone_ws/src/unitree_legged_sdk/lib/libunitree_legged_sdk_amd64.so when searching for -lunitree_legged_sdk_amd64
/usr/bin/ld: cannot find -lunitree_legged_sdk_amd64
collect2: error: ld returned 1 exit status
make[2]: *** [unitree_ros_to_real/unitree_legged_real/CMakeFiles/position_lcm.dir/build.make:101: /home/parallels/capstone/capstone_ws/devel/lib/unitree_legged_real/position_lcm] Error 1
make[1]: *** [CMakeFiles/Makefile2:2753: unitree_ros_to_real/unitree_legged_real/CMakeFiles/position_lcm.dir/all] Error 2
/usr/bin/ld: skipping incompatible /home/parallels/capstone/capstone_ws/src/unitree_legged_sdk/lib/libunitree_legged_sdk_amd64.so when searching for -lunitree_legged_sdk_amd64
/usr/bin/ld: cannot find -lunitree_legged_sdk_amd64
collect2: error: ld returned 1 exit status
make[2]: *** [unitree_ros_to_real/unitree_legged_real/CMakeFiles/walk_lcm.dir/build.make:101: /home/parallels/capstone/capstone_ws/devel/lib/unitree_legged_real/walk_lcm] Error 1
make[1]: *** [CMakeFiles/Makefile2:2641: unitree_ros_to_real/unitree_legged_real/CMakeFiles/walk_lcm.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j4 -l4" failed
```
What I have done is following [this](https://github.com/unitreerobotics/unitree_ros/files/9492749/ROS.Build.instructions.for.Ubuntu.pdf) pdf tutorial from @andrewhstovell 

Can someone help me please ?

andrewhstovell, 2022-10-26:
```
skipping incompatible /home/parallels/capstone/capstone_ws/src/unitree_legged_sdk/lib/libunitree_legged_sdk_amd64.so when searching for -lunitree_legged_sdk_amd64
```
Is your machine ARM based? If so, in `unitree_legged_sdk/CMakeFiles.txt` change line 10

FROM: 
`set(EXTRA_LIBS -pthread libunitree_legged_sdk_amd64.so lcm)`

TO: 
`set(EXTRA_LIBS -pthread libunitree_legged_sdk_arm64.so lcm)`

<hr/>

_Going off on a tangent_, i've learned that my tutorial is very incomplete. I will make a point to update it, perhaps even put out a getting started video.

My tutorial focuses on `unitree_ros_to_real v3.4.0` in combination with `unitree_legged_sdk v3.4.2`

However, i've since learned that more recently made Go1 robots likely aren't compatible with this SDK version, as they have newer software installed (sport mode). So make sure to check your robot is compatible with `unitree_legged_sdk v3.4.2` either by connecting to it via wifi and checking the currently installed software and firmware against the ReadME, or better yet, by running the `example_walk` after building. Assuming that you've set up your ethernet connection correctly and have been able to ping `192.168.123.161` successfully, the quadruped should perform a range of movements, if nothing happens, its likely to be a compatability thing. 

`unitree_ros_to_real` has recently had some bug fixes for `v3.5.1` and has released `v3.8.0` (compatible with `unitree_legged_sdk` of same versions) so, when i do re-write the tutorial i'll be sure to include instructions depending on the combination people need/want to use, and ill also be sure in include LCM and CMake build instructions, along with connection instructions which are currently missing.  :)

JulienRineau, 2022-10-26:
Thanks ! 
I've encountered several other problems. Ive summarized everything I've done to make it works on [this](https://drive.google.com/file/d/1W7t7Y_xllz5-MHq60X3pE-XEJg6Dg9gU/view?usp=sharing) pdf tutorial based on yours. Hope it can help others