unitreerobotics/unitree_ros #55: has no member named ‘eeForce’

tcr1314, 2022-10-28:
error: ‘unitree_legged_msgs::LowState {aka struct unitree_legged_msgs::LowState_<std::allocator<void> >}’ has no member named ‘eeForce’; did you mean ‘footForce’?
         lowState.eeForce[0].x = msg.wrench.force.x;
How to solve this problem?

tcr1314, 2022-10-28:
[image]

JonasFovea, 2022-12-02:
Hi!

I think [this video](https://youtu.be/R4dW1xf1Y6A?t=97) might help :)

xyq1223, 2023-03-21:
Hello,this is caused by the wrong version of unitree_legged_msgs.
There are two solutions for this.

Method 1: You can choose to put the latest unitree_legged_msgs from the unitree_ros_to_real (v3.2.1） package under ~/catkin_ws/src and recompile it.
https://github.com/unitreerobotics/unitree_ros_to_real/tree/v3.2.1

Method 2: Add the following statements to the unitree_legged_msgs/msg/LowState.msg file and recompile.

Cartesian[4] eeForceRaw
Cartesian[4] eeForce          #it's a 1-DOF force infact, but we use 3-DOF here just for visualization