unitreerobotics/unitree_ros #57: Robot slides by himself

JulienRineau, 2022-11-29:
When running `roslaunch unitree_gazebo normal.launch rname:=go1 wname:=stairs` the robot slides on the ground even without doing any movements.
Is there any way to fix this ?

jmsalash, 2022-12-22:
I think this has to do with the simulated wind in gazebo. Disable it and it should go away

aatb-ch, 2023-01-10:
it is, can confirm, somehow the scene has wind enabled.

MrGriv, 2023-02-15:
it doesn't work.

https://user-images.githubusercontent.com/115357430/219042291-86e81f2d-5492-4495-bf95-b2462e4fa0bf.mp4

MrGriv, 2023-02-15:
https://user-images.githubusercontent.com/115357430/219046257-bfb59538-2b6a-43f0-8246-81ca6b25c815.mp4

MrGriv, 2023-02-16:
It doesn't work
https://youtu.be/1DTEu-J1l0A

Sergim96, 2023-03-30:
I experienced a similar issue when running the B1 simulation.

I managed to solve it by removing the [calf rotor collision](https://github.com/unitreerobotics/unitree_ros/blob/26ceaa85d0725c4f849272840d0eca9b6ebcd01c/robots/b1_description/xacro/leg.xacro#L225-L230) in the leg xacro, apparently it was causing internal collisions and resulting in an unstable simulation.

I think this is the same issue you might be experiencing in Go1.