unitreerobotics/unitree_ros #60: Control Go1 turn head up or head down while walking?

quannguyen123456, 2022-12-11:
Hi Every One

I try to control Go1 Unitree by other outside PC, send High level control command by ethernet gate.
Every thing is Ok , but i want to control robot turn head up or down while it is walking.
what should i do? i tink it is possible because we can control it by its remote control.

thank you very much .

JonasFovea, 2022-12-16:
Have you tried changing the tree `euler` values?

This issue is probably something for the SDKs repository, rather than this one...

quannguyen123456, 2022-12-23:
Thank for your answer JonasFovea.
With Mode 1, i could use  euler parametter  to control robot head up or down, but Mode 1 mean robot can not move. 
I need Robot work on Mode 2(Robot can move in this mode), but still head up or down. Use including Remote control, i can easily do it.
I think i use SDKs 3.8, not newest.

JonasFovea, 2022-12-23:
I believe it worked for me with version 3.5 but I'm not sure.

You can maybe try that.

The newer versions of the SDK are for different robots, so be careful to select a supported version

quannguyen123456, 2023-01-04:
Thanyou very much. I will try with version 3.5.