unitreerobotics/unitree_ros #66: Wrong parameters with B1 gazebo

zzhou387, 2023-04-08:
Hi,

Some issues I found with B1 simulation in gazebo:
1. The namespace is wrong: https://github.com/unitreerobotics/unitree_ros/blob/26ceaa85d0725c4f849272840d0eca9b6ebcd01c/robots/b1_description/xacro/gazebo.xacro#L6 It's supposed to be `b1_gazebo`
2. The joint limits seem to be very wrong:
https://github.com/unitreerobotics/unitree_ros/blob/26ceaa85d0725c4f849272840d0eca9b6ebcd01c/robots/b1_description/xacro/const.xacro#L73 The peek torque is reported to be 210 Nm on the website.
3. For some reason, the estimated torque from gazebo is always clamped to be under 55 Nm. I found this magic number coming from the aliengo specs. However, I'm not sure where to change this magic number since it's clearly not loaded from the xacro or urdf. Here's a screenshot I took when I was trying to send a command higher than 55 Nm.
[image]

These should all be easy fix and thanks in advance.

xiaoliangstd, 2023-04-29:
hello, 
1.  for the first question:  we have corrected the <robotNamespace> micro problem of b1_description in the newest push .
2.  The peak torque is reported to be 140 Nm on the website, which corresponds correctly to the URDF.
3.  we are checking this problem.

zzhou387, 2023-05-31:
> hello,
> 
> 1. for the first question:  we have corrected the  micro problem of b1_description in the newest push .
> 2. The peak torque is reported to be 140 Nm on the website, which corresponds correctly to the URDF.
> 3. we are checking this problem.

Thanks. Any updates on the third problem?

zhthree3, 2023-08-14:
> @zzhou387 Hello, there may be a problem with the emulator settings, because the peak torque in our urdf file is 140 Nm, we have discussed it with our R&D colleagues, and we will give you an answer within a week.

Any updates?

zzhou387, 2023-09-19:
> @zzhou387 Hello, there may be a problem with the emulator settings, because the peak torque in our urdf file is 140 Nm, we have discussed it with our R&D colleagues, and we will give you an answer within a week.

Hello, it has been a while. Could you provide some updates on this?