unitreerobotics/unitree_ros #69: URDF incomplete

Sergim96, 2023-05-22:
Hi,

The URDF of B1 is incomplete, we have identified the following missing components and issues:
- The IMU link is not properly positioned, instead it is just placed aligned with the trunk.
- There is no information about where the 5 Realsense cameras are placed within the robot body.
- The collision models of some limbs create internal collisions which result in unstable simulations, as reported in https://github.com/unitreerobotics/unitree_ros/issues/57
- The total mass of the robot does not match the weight of the real robot.

Are there any plans to update the URDF?

xyq1223, 2023-05-24:
Hello, thank you very much for your correction, I have already reported this problem to our R&D colleagues, please look forward to our follow-up updates

AlessiaLiNoce, 2023-08-27:
Hi, I'm also start to working with the B1. The urdf has been fixed or still have the same issues?