unitreerobotics/unitree_ros #74: YOLO darknet_ros to real robot Unitree Go1

AlbiBechis, 2023-06-29:
Hello everyone,
I would like to add YOLO feature to Unitree Go1.
I'm quite new to ROS and Unitree but I'm trying to understand where and how to put https://github.com/leggedrobotics/darknet_ros in Go1 after ssh.
In the raspberry I can see catkin_ws & ros_catkin_ws but I'm unable to catkin_make, do someone have some suggestions? 

Thank you so much

AlbiBechis, 2023-06-30:
I understand to use darknet_ros from my own ubuntu terminal, but the result is: "Waiting for image..." in loop.
I checked camera topics and works well, but why I cannot see result from unitree cameras? Thanks

xyq1223, 2023-08-03:
Hello,

1. The camera topic is on the bus. Raspberry Pie is just a data transfer station in the go1 robot dog. You can read related topics by reading the ros topic on the bus.
2. If it involves equipment occupation (for example, if a project wants to call the camera, but it shows that the camera is occupied, it is necessary to manually close the process of the camera)