unitreerobotics/unitree_ros #79: Any guides for additional ROS packages for Go1 Edu/ resetting to factory defaults settings?

JieLinBeans, 2023-08-02:
Hello, 
I realised the recommended system to use for these packages are Ubuntu 18.04, but it seems like the system on the Go1 Edu is Raspbian Buster, is there anyway or any guides available to 
1) Install additional ros packages on raspbian buster - the default sudo apt-get installs points to ros-noetic
2) Resetting the system to factory/ ros packages to factory defaults?
Please advise. 

Thank you!

xyq1223, 2023-08-07:
Hello,the question you mentioned is about how to connect the Raspberry Pi to the external network. Our suggestion for you is to connect the external USB network card to the Internet through the USB port of the board.

JieLinBeans, 2023-08-07:
Sorry, I think I might not have phrased clearly. I actually do have internet access on the Go1, but I am not able to install the correct packages required for ROS Melodic. I meant to ask about: 
1) Installation procedures of additional ros packages. Currently I am trying to run _sudo apt-get install ros-melodic-<packagename>_, I am connected to the internet, but I am only able to install ros noetic packages. Is there a reference guide or specific way that you use to install melodic packages, using rosinstall? 
2) Reset rpi to factory default. For this, I just want to check if there is anyway to reset the rpi, in the event that the rpi system is messed up in anyway after making modifications.

Thank you!

xyq1223, 2023-09-26:
Hello, regarding the first question, we do not recommend changing the ros version. Regarding the second question, currently the RPI of our Go1 robot dog cannot be restored to factory settings. It is recommended that you use NANO or NX when debugging the program.