unitreerobotics/unitree_ros #85: Could not load class 'unitree_legged_control/UnitreeJointController'

superhanye, 2023-11-25:
[INFO] [1700903912.720351, 0.000000]: Loading controller: FL_hip_controller
[ERROR] [1700903912.747374154, 0.004000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903912.747811660, 0.004000000]: Could not load controller 'FL_hip_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903912.761173875, 0.004000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903913.767087, 0.297000]: Failed to load FL_hip_controller
[INFO] [1700903913.770062, 0.297000]: Loading controller: FL_thigh_controller
[ERROR] [1700903913.778097836, 0.297000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903913.778401354, 0.297000000]: Could not load controller 'FL_thigh_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903913.778564389, 0.297000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903914.780696, 0.585000]: Failed to load FL_thigh_controller
[INFO] [1700903914.783390, 0.585000]: Loading controller: FL_calf_controller
[ERROR] [1700903914.789590854, 0.586000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903914.789748757, 0.586000000]: Could not load controller 'FL_calf_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903914.789821986, 0.586000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903915.791242, 0.857000]: Failed to load FL_calf_controller
[INFO] [1700903915.793823, 0.857000]: Loading controller: FR_hip_controller
[ERROR] [1700903915.803077958, 0.858000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903915.803491377, 0.858000000]: Could not load controller 'FR_hip_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903915.803566560, 0.858000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903916.805549, 1.115000]: Failed to load FR_hip_controller
[INFO] [1700903916.808772, 1.115000]: Loading controller: FR_thigh_controller
[ERROR] [1700903916.818865601, 1.115000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903916.819258731, 1.115000000]: Could not load controller 'FR_thigh_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903916.819308202, 1.115000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903917.834368, 1.356000]: Failed to load FR_thigh_controller
[INFO] [1700903917.837156, 1.356000]: Loading controller: FR_calf_controller
[ERROR] [1700903917.852258798, 1.357000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903917.852427184, 1.357000000]: Could not load controller 'FR_calf_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903917.852499369, 1.357000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903918.854287, 1.625000]: Failed to load FR_calf_controller
[INFO] [1700903918.856858, 1.625000]: Loading controller: RL_hip_controller
[ERROR] [1700903918.865789685, 1.626000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903918.865933443, 1.626000000]: Could not load controller 'RL_hip_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903918.866006404, 1.626000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903919.867094, 1.882000]: Failed to load RL_hip_controller
[INFO] [1700903919.870861, 1.882000]: Loading controller: RL_thigh_controller
[ERROR] [1700903919.883966888, 1.882000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903919.884358629, 1.882000000]: Could not load controller 'RL_thigh_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903919.884437806, 1.882000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903920.886209, 2.203000]: Failed to load RL_thigh_controller
[INFO] [1700903920.888829, 2.204000]: Loading controller: RL_calf_controller
[ERROR] [1700903920.895657657, 2.204000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903920.895793520, 2.204000000]: Could not load controller 'RL_calf_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903920.895872741, 2.204000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903921.899814, 2.500000]: Failed to load RL_calf_controller
[INFO] [1700903921.902412, 2.500000]: Loading controller: RR_hip_controller
[ERROR] [1700903921.908850735, 2.500000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903921.909126180, 2.500000000]: Could not load controller 'RR_hip_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903921.909200954, 2.500000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903922.910565, 2.897000]: Failed to load RR_hip_controller
[INFO] [1700903922.913190, 2.897000]: Loading controller: RR_thigh_controller
[ERROR] [1700903922.921217576, 2.897000000]: Could not load class 'unitree_legged_control/UnitreeJointController': Could not find library corresponding to plugin unitree_legged_control/UnitreeJointController. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903922.923458180, 2.897000000]: Could not load controller 'RR_thigh_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[... cut ...] description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1700903923.943228648, 3.259000000]: Could not load controller 'RR_calf_controller' because controller type 'unitree_legged_control/UnitreeJointController' does not exist.
[ERROR] [1700903923.943385075, 3.259000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1700903924.945369, 3.661000]: Failed to load RR_calf_controller

How to solve it?

fanyahao1, 2024-01-05:
I got the same issue, have you solved it?

glianghan, 2025-04-15:
I got the same issue

TAT2023, 2025-06-27:
same issue, have you solved it?