unitreerobotics/unitree_ros #91: Unitree Go2 missing camera, lidar

dhanishmeht, 2024-03-14:
Hello,

The Unitree Go2 URDF is missing the joints/links to the built-in camera and lidar.
Also is there plans to include the SLAM navigation computer along with the radar and depth camera in the URDF

yiyuc, 2024-03-15:
The built-in lidar is in the go2_description.urdf file under urdf folder. In the last few lines you will see the link for lidar. Not sure about the other stuff though

dhanishmeht, 2024-04-23:
The robot came with the MID-360 radar and the Intel RealSense D435i camera, here is the link from Unitree docs,
https://support.unitree.com/home/en/developer/SLAM%20and%20Navigation_service
I was wondering if these can be added to the URDF

xiaoliangstd, 2024-06-26:
Hello, because the standard version of the Go2 does not include the aforementioned devices, they are not present in the URDF. You could contact our technical support team at support@unitree.com.

xiaoliangstd, 2024-06-26:
https://www.unitree.com/en/connect/