unitreerobotics/unitree_ros #93: catkin_make error about plugin/draw_force_plugin.cc.o

abloomyt, 2024-04-13:
ubuntu version : 20.04
ros version : noetic
gazebo version : 11
I'm sorry to bother you. I met some problems when compiling unitree_gazebo, as shown in the picture below. Could someone please answer them for me? Thank you!
[image]

yiyuc, 2024-04-24:
Everything still runs if you don't compile with the plugins. When updating from melodic to noetic, some functions in the visualization part change so you can't compile with ros noetic

abloomyt, 2024-04-27:
I reinstalled the system and the compilation passed. Thank you for your help.

closed 2024-04-27 by abloomyt