unitreerobotics/unitree_ros #100: xacro for H1 and G1

anahrendra, 2024-06-14:
Will you update the xacro files for H1 and G1? It's currently missing in the repo.

Thanks!

xiaoliangstd, 2024-06-25:
I'll update these today or tomorrow.

bhyu0924, 2024-06-27:
@xiaoliangstd 
I want to let you know that someone is dying because of the wait.. Help!

xiaoliangstd, 2024-06-27:
> @xiaoliangstd I want to let you know that someone is dying because of the wait.. Help!
I have already created an initial version of the xacro for g1. What is your contact information? WhatsApp?

bhyu0924, 2024-06-27:
> > @xiaoliangstd I want to let you know that someone is dying because of the wait.. Help!
> > I have already created an initial version of the xacro for g1. What is your contact information? WhatsApp?

I would like to inform you that I am not a user of WhatsApp. You can contact me directly via email at bhyu@kaist.ac.kr.
I am very pleased to hear that you have already created the Xacro file for G1. I look forward to receiving the Xacro file for H1 soon.

Thank you for your attention to this matter.

xiaoliangstd, 2024-06-27:
> > > @xiaoliangstd I want to let you know that someone is dying because of the wait.. Help!
> > > I have already created an initial version of the xacro for g1. What is your contact information? WhatsApp?
> 
> I would like to inform you that I am not a user of WhatsApp. You can contact me directly via email at [bhyu@kaist.ac.kr](mailto:bhyu@kaist.ac.kr). I am very pleased to hear that you have already created the Xacro file for G1. I look forward to receiving the Xacro file for H1 too soon.
> 
> Thank you for your attention to this matter.
I have already sent you the email, and I will do the rest as soon as possible. Some other work is pushing me。

xiaoliangstd, 2024-06-27:
I just updated a new one g1 xacro files.

PocaBBQ, 2024-09-20:
> I just updated a new one g1 xacro files.

do you have xacro files for H1 version?