unitreerobotics/unitree_ros #112: The URDF problem of B2W

hanzhi0410, 2024-10-24:
When I use Isaac Gym to load the URDF of BW2, I find that its right wheel always gets stuck in the ground.
[image]
[image]

hanzhi0410, 2024-10-24:
[image]
I found that this seems to be caused by an incorrect collision body.

dongjukim123, 2025-03-18:
> I found same issue Did you solve it?

I solved that problem editing the collisioin mesh related to the thigh and calf rotating 2/pi roll