unitreerobotics/unitree_ros #120: Zhexi Oray Tunnel Backdoor in Go1 has been disclosed please advise...

ZhexiBackdoor, 2025-03-21:
https://x.com/Bin4ryDigit/status/1903092582021804543

https://think-awesome.com/download_unitree_report

Tunnel services
CloudSail
CloudSail (Zhexi) is a remote access tunnel service developed by Zhexi Technology, primarily
targeted at Chinese markets. The service is designed to provide NAT traversal and remote access
capabilities for IoT devices, industrial equipment, and other networked systems. While the service
itself is a legitimate tool for remote device management. It can be compared to ngrok, cloudflare
tunnel etc.
To understand the CloudSail service and its capabilities more read the FAQ:
https://jmz.zhexi.tech/faq/
What can the service do?
The CloudSail service can establish a connection from any device to another, even across
diﬀerent networks, depending on your configuration.
For example, you could open a TCP connection to a connected device: the client on the device
connects to the CloudSail network, allowing you to route connections to services running on that
device.
This means if an SSH daemon is running on the client device, you could connect to it through
CloudSail, even if the local network blocks incoming connections or lacks port forwarding —
eﬀectively circumventing NAT and firewall restrictions.
This can be especially useful when your device is on a mobile network with CGNAT or similar
configurations, which would typically prevent external access. However, this level of access can
also be dangerous. The decision to enable such functionality should always remain with the user,
not the manufacturer.

ZhexiBackdoor, 2025-03-23:
I can see you killed the token for the zhexi backdoor and removed the API key. Now you have to make public disclosure. Don’t forget that part. You are welcome for the email, you cold have at least said thanks. "2025/03/23 15:11:47 2926 ERROR [worker:handleRegisterResponse] register failed. Error:token not valid"

[image]

PPPDUD, 2025-04-13:
This is a very lazy way of concealing a backdoor and API keys.
I can think of 1000 better ways of doing this _off the top of my head_.

I looked through the codebase, and it gives off a very amateur, hobbyist feel.
For a multi-thousand dollar device, this could use some serious work.